ouster-lidar/ouster-sdk #627: Using SLAM for Real-Time Pose Matrix Extraction

moonmoonmoonmoon, 2024-11-19:
Hello, I would like to use SLAM to obtain the pose matrix of each frame in real-time collected data. Could you provide a function similar to `process_pcap` (which uses SLAM to retrieve the pose matrix of each frame in a recorded PCAP file)?

def process_pcap(pcap_path, output_dir, voxel_size=None):
   
    
    os.makedirs(output_dir, exist_ok=True)

   
    data_source = open_source(pcap_path)
    slam = KissBackend(data_source.metadata, max_range=75, min_range=1, voxel_size=1.0)

 
    scans_w_poses = map(partial(slam.update), data_source)

    for idx, scan_w_pose in enumerate(scans_w_poses):
        
        col = client.first_valid_column(scan_w_pose)

       
        scan_pose = scan_w_pose.pose[col]  
        scan_ts = scan_w_pose.timestamp[col]

        
        base_filename = f"frame_{idx:06d}"
        pcd_path = os.path.join(output_dir, f"{base_filename}.pcd")
        pose_path = os.path.join(output_dir, f"{base_filename}_pose.json")
        timestamp_path = os.path.join(output_dir, f"{base_filename}_timestamp.txt")

       
         pcd = save_scan_to_pcd(data_source.metadata,scan_w_pose, scan_pose, pcd_path, voxel_size)

       
        pose_data = {
            "transform_matrix": scan_pose.tolist(),
            "frame_id": idx,
            "timestamp": float(scan_ts)  
        }

Samahu, 2024-11-19:
Not sure if I understood your question correctly but the `open_source` can also handle sensor_url or osf files.

moonmoonmoonmoon, 2024-11-22:
Thank you very much. I notice you update the version, now I use your example code in mapping part in ouster document, it works.

closed 2024-11-22 by Samahu