ouster-lidar/ouster-sdk #633: How to Assign Different IP Addresses to Two LiDAR Sensors Connected via Ethernet?

moonmoonmoonmoon, 2024-12-20:
I am using a host machine connected to two LiDAR sensors via two Ethernet cables. When I use the command ping xxx (where xxx is the serial number of the sensor), I discovered that both LiDAR sensors have the same IP address. As a result, I am unable to receive data properly.  Is there a way to assign different IP addresses to these two sensors? Any guidance or suggestions would be greatly appreciated!

matthew-lidar, 2024-12-23:
You can use this endpoint to set the static IP of the sensor: https://static.ouster.dev/sensor-docs/image_route1/image_route2/common_sections/API/http-api-v1.html#put-api-v1-system-network-ipv4-override

moonmoonmoonmoon, 2024-12-24:
> You can use this endpoint to set the static IP of the sensor: https://static.ouster.dev/sensor-docs/image_route1/image_route2/common_sections/API/http-api-v1.html#put-api-v1-system-network-ipv4-override

thank you!