ouster-lidar/ouster-sdk #636: Sensor utils stream info gives negative min and max timestamps

hirenpatel1207, 2025-01-07:
**Describe the bug**
I am trying to obtain the min and max timestamps in the pcap recorded file but the stream_info gives out negative timestamps.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
- Ouster Sensor? OS-1
- Programming Language?C++ Python
- Operating System?  Windows
- Machine Architecture? x64

Basically, I want to get the progress while reading the lidar data from the pcap file. For that I first need the min and max timestamp present in the pcap file. With that information, I can then calculate the progress percentage. (Correct me if my approach is incorrect)
Here is the sudo code that I used to get the min max timestamps.

`   std::shared_ptr<ouster::sensor_utils::stream_info> streamInfo = ouster::sensor_utils::get_stream_info(
                                                                  "path to pcap file"
                                                                      );
     std::cout<< (double)streamInfo->timestamp_min.count() << std::endl;
     std::cout<< (double)streamInfo->timestamp_max.count() << std::endl; `
     
     Both of these give out negative values, I even tried the python sdk, but the values I got are same. I can share a sample pcap file if required.

      Also, to then calculate progress, I am using the packet_info like:
      `ouster::sensor_utils::packet_info packet_info;
     if ( ouster::sensor_utils::next_packet_info(*_pcapHandle, packet_info))
     {
     double currentScanTime = (double) packet_info.timestamp.count();
double progress = (currentScanTime - timestamp_min)  / (timestamp_max- timestamp_min);
     }
      `
      Is this correct way?

twslankard, 2025-01-07:
@hirenpatel1207 thanks for your report.

I'm not able to reproduce the issue with the latest master commit using test data I have on hand.

Can you please tell me which version of the SDK you are using and how you recorded the PCAP?

hirenpatel1207, 2025-01-14:
Hi @twslankard
The sdk version was release-0.13.0 
I had used the example code provided at https://github.com/ouster-lidar/ouster-sdk/issues/611#issuecomment-2339041390

Thanks,
Hiren

hirenpatel1207, 2025-01-14:
Also, what is the better way to store and read lidar data (outside ROS environment), pcap or osf file? In the docs it says the osf format has some advantages, but on the other hand replaying an osf file in the Ouster Studio is not possible right? Also the ouster-ros does support replaying a PCAP file but not an osf file.

twslankard, 2025-01-14:
@hirenpatel1207 

You're correct; Ouster Studio and ouster-ros only support PCAP playback at present.

I will try to replicate the timestamp issue you're seeing with the example code you've provided for recording.

Cheers,

Tom Slankard
Staff Engineer
Ouster Inc.