ouster-lidar/ouster-sdk #639: How to save IMU stream in osf files

joaocalen, 2025-01-17:
I'm creating a project using lidar which I log into a .osf every scan given by the lidar sensor. I found this easy using this method of class Writer:

void save(uint32_t stream_index, const LidarScan& scan,
              const ouster::osf::ts_t timestamp);

But how can I save also IMU data? I have to use the save_message to store an array of bytes with the IMU data or is there a better way to do it?

Samahu, 2025-01-17:
Unfortunately OSF currently doesn't support IMU data yet.
If you need to preserve IMU use pcap or bag file format and if you are using the ouster-cli use the `save_raw` command.

Samahu, 2025-01-21:
@joaocalen you can try using the `SCAN_FIELD`  of the LidarScan to include the IMU scan as part of lidar-data. However, you will be responsible for the most of encoding/decoding the IMU data. Here is a python example:

```
import numpy as np
from ouster.sdk.client import FieldClass, LidarScan

NUM_IMU_MEASURMENTS_PER_SCAN = 100

scan = LidarScan(128, 1024) # or obtain the lidar scan from some place
scan.add_field('imu', np.uint8, (NUM_IMU_MEASURMENTS_PER_SCAN, 6), FieldClass.SCAN_FIELD)

for i, (linear_acc, angular_vel) in zip(range(NUM_IMU_MEASURMENTS_PER_SCAN), imu_source):
  scan.field('imu')[i][0] = linear_acc.x
  scan.field('imu')[i][1] = linear_acc.y
  scan.field('imu')[i][2] = linear_acc.z
  scan.field('imu')[i][3] = angular_vel.x
  scan.field('imu')[i][4] = angular_vel.y
  scan.field('imu')[i][5] = angular_vel.z

writer = osf.Writer(osf_file, source.metadata)
writer.save(0, scan, scan.packet_timestamp)
...
```

closed 2025-01-27 by Samahu