ouster-lidar/ouster-sdk #642: ouster- debug build

LEECHANGJOON0324, 2025-02-12:
Hello,
I would like to build ouster-sdk in Debug mode and run tests, but I am encountering an error.

OS: Windows 10
IDE: Visual Studio 10, C++
Build System: vcpkg, CMake

It works fine in Release mode, but when running in Debug mode, an error occurs in sniffer.cpp at the snifferConfiguration and set_filter(const std::string& filter) functions.

The std::string type is recognized as <error-type>. And the filter string is assigned garbage values. 
As a result, memory allocation for the impl->pcap_reader variable in pcap.cpp becomes impossible.

What could be the issue?

- Debug mode

[image]

- Release mode

[image]

chrisbayruns, 2025-02-18:
Short Answer: 
The MSVC compiled binary ABIs can not be mixed, you need to make sure all libraries that are compiled follow the same build type (release/debug).

Long Answer:
Much like how software libraries have an API(Application Programmer Interface) that allows you to specify how to call things, compiled binaries have an ABI(Application Binary Interface). This ABI basically denotes basic structural interfaces between different compiled binaries. The ABI produced by the MSVC compiler is different between release and debug. This means that every single library you use needs to have the same "build type" other wise you run into stuff like this. In this case the std::string structure is not the same on debug as it is on release.

LEECHANGJOON0324, 2025-03-05:
According to the example page, it says to build in Release mode.

Does this mean that ouster-sdk does not officially support Debug mode?

Thank you.

[image]

matthew-lidar, 2025-03-05:
The SDK supports being built/used in either, just may run quite slowly in debug mode due to the lack of optimizations. 

You just need to make sure all of your programs/dependencies are built in the same mode Debug vs Release or they will break inexplicably.

Samahu, 2025-03-05:
You can definitely build the software in debug mode, however, as Matthew noted some methods in SDK will run extremely slow when using debug build which has implications over packets processing and batching.

That being said we are are implementing some improvements for 0.15 that should make it more feasible to run the SDK even when using debug builds.