ouster-lidar/ouster-sdk #643: HTTP request fails intermittently, but works via command line curl

themightyoarfish, 2025-02-20:
https://github.com/ouster-lidar/ouster-sdk/blob/225053dc683ca9a5f3d843ef4a5099ce9eee7a68/ouster_client/src/sensor_http.cpp#L18

This line in the client sometimes fails for us with this error

```
  what():  CurlClient::execute_request failed for the url: [http://os-122422000616.local/api/v1/system/firmware] with the error message: Couldn't connect to server
```

When `SensorHTTP::create()` is invoked. Simply retrying a few times makes it go away. Meanwhile I can never reproduce this behaviour when just using curl on the command line in a loop

```
while true; do curl --request GET http://os-122422000616.local/api/v1/system/firmware && echo "result was $?"; done
```

(always successful).

This has happened with several different rev7 devices, and I'm not sure how to troubleshoot. The scanners work normally once connected.

Any advice on how to investigate?

I have not yes tried with current master, as it has introduced some install issues wrt to the version we're using, but the HTTP code is identical.

themightyoarfish, 2025-04-07:
Any help here? Am i the only one experiencing this=

matthew-lidar, 2025-04-07:
What timeout_sec value are you using? We have found that some systems or network configurations need a timeout of at least a few seconds.

themightyoarfish, 2025-04-07:
We are using a 10 second timeout, but it terminates right away, without waiting for 10 seconds as far as i can tell