ouster-lidar/ouster-sdk #649: SensorScanSource connection to multiple lidars

jjoepayne, 2025-03-17:
I am currently running into an issue with the SensorScanSource class where I cannot get scan information from any sensors beyond the first 3.

When using the client_example, I can connect to all 6 of the OS1s that I am trying to use (I will get the sensor information output with serial numbers, etc. from them all), however when it goes into the loop of collecting scans, it will only ever return scans from indices 0,1 and 2.

It even works if I simultaneously run the example on the first 3 ips and the last 3 ips at once.

Is there something I'm missing here or is there anything I can try to fix this? 

**Platform:**
- Ouster Sensor: OS-1
- Ouster Firmware Version: 3.1
- Programming Language: C++
- Operating System: Ubuntu 22.04.5
- Machine Architecture? x86-64

matthew-lidar, 2025-03-17:
I'm not sure we've tried it with more than 3 sensors at a time. I'll take a look to see if we can reproduce this.

matthew-lidar, 2025-03-17:
Do you know what lidar mode (e.g. 1024x10 2048x20) beam counts (e.g. 64 128) and packet formats (e.g. RNG15_REFL8_NIR8) you were using with these sensors?

I've quickly tried this out with 4 sensors through our Python bindings of that API and don't see any issues.

jjoepayne, 2025-03-17:
I am using 1024x128 and it looks like they are currently set to RNG19_RFL8_SIG16_NIR16