ouster-lidar/ouster-sdk #650: Live SensorScanSource while simultaneously logging data to .pcap

loughlindd, 2025-04-09:
Hi 
Thanks for the development of this SDK. I have a small question (using SDK from python)

**Describe your question**
I am working with live sensors, and would like to log the data live to .pcap files, while at the same time processing the LidarScans data. 
For this: 
- sensor.SensorScanSource allows to iterate over the live scans, without the ability to log the data to pcap files. 
- client.SensorPacketSource allows to iterate over the packets, but not convert a group of packets with the same measurement_id to LidarScan. 
- I see the Scans deprecated object, which wan instantiate a Scans object from a PacketSource. But it is deprecated. 

Is there a way to log live data live while accessing LidarScan objects ? Or do I have to use the deprecated Scans object ?
Thanks

**Platform (please complete the following information):**
Using ouster-sdk in python 3.10 on Windows

Samahu, 2025-04-09:
You can do so by implementing your own **ScanSource** on top of the **PacketSource** and then process **LidarPacket** to produce **LidarScan** while simultaneously saving the packets to a file, here is an example:

```
import ouster.sdk.client as client

SIDX = 0  # show casing the example for a single sensor for simplicity 

class CustomScanSource:

    def __init__(self, source_url):
        self._source = client.SensorPacketSource(source_url)
        self._metadata = self._source._metadata
        self._writer = open("file.pcap")

    def __iter__(self):
        packet_format = core.PacketFormat(self._metadata[SIDX])
        field_types = core.get_field_types(self._metadata[SIDX])
        columns_per_packet = self._metadata[SIDX].format.columns_per_packet
        W = self._metadata[SIDX].format.columns_per_frame
        H = self._metadata[SIDX].format.pixels_per_column
        batcher = core.ScanBatcher(self._metadata[SIDX])
        ls_write = LidarScan(H, W, field_types, columns_per_packet)

        for idx, packet in self._source:
            if isinstance(packet, core.LidarPacket):
                if batcher(packet, ls_write):
                    yield [copy.deepcopy(ls_write)]

            // serialize the pacekt to a pcap
            serialized_packet = serialize(packet)
            self._writer.write(serialized_packet)

    def close()
            self._source.close()
            self._writer.close()
```

loughlindd, 2025-04-10:
Hi, 
Thanks very much for your help. I am now faced with another issue. 
I have adapted your code to have a logrotate to a new .pcap file every 10 seconds. I am however not sure about how to save the data in the pcap file. You suggest using serialize() in your code, which I cannot seem to find as a pre-defined function in ouster-sdk. 
I tried saving with packet.buf.tobytes(), as in the code below. 
However, when I 

```
from contextlib import closing
import numpy as np
from ouster.sdk import client, sensor
import time
import matplotlib.pyplot as plt
import ouster.sdk.client as client
import copy
import os

def round_time_to_nearest_10s(timestamp: float) -> float:
    """Round a timestamp to the lowest 10 seconds."""
    return int(timestamp / 10) * 10

SIDX = 0  # show casing the example for a single sensor for simplicity 
class CustomScanSource:
    def __init__(self, hostname, config, save_dir, metadata=None):
        self._source = client.SensorPacketSource([(hostname, config)])
        self._metadata = self._source._metadata
        self._save_dir = save_dir
        self._tstart = round_time_to_nearest_10s(time.time())
        self._tlastlog = self._tstart
        self._writer = open(f"{self._save_dir}/{self._tstart}.pcap", "wb")

        self.save_metadata()

    def save_metadata(self):
        metadata_path = os.path.join(self._save_dir, f"{self._tstart}.json")
        with open(metadata_path, "w") as f:
            f.write(self._metadata[SIDX].to_json_string())

    def rotate_writer(self):
        self._writer.close()
        tlog = round_time_to_nearest_10s(time.time())
        self._writer = open(f"{self._save_dir}/{tlog}.pcap", "wb")
        self._tlastlog = tlog
        print("Rotating writer to new file:", self._writer.name)

    def __iter__(self):
        packet_format = client.PacketFormat(self._metadata[SIDX])
        field_types = client.get_field_types(self._metadata[SIDX])
        columns_per_packet = self._metadata[SIDX].format.columns_per_packet
        W = self._metadata[SIDX].format.columns_per_frame
        H = self._metadata[SIDX].format.pixels_per_column
        batcher = client.ScanBatcher(self._metadata[SIDX])
        ls_write = client.LidarScan(H, W, field_types, columns_per_packet)

        for _, packet in self._source:
            if isinstance(packet, client.LidarPacket):
                if batcher(packet, ls_write):
                    yield [copy.deepcopy(ls_write)]

            if time.time() - self._tlastlog > 10:
                self.rotate_writer()

            self._writer.write(packet.buf.tobytes())

    def close(self):
            self._source.close()
            self._writer.close()

if __name__ == "__main__":
    
    hostname = "192.168.100.20"
    config = client.SensorConfig()
    config.udp_port_lidar = 7520
    config.udp_port_imu = 7521

    with closing(CustomScanSource(hostname, config, "C:/Users/lough/Desktop/")) as source:
        
        for scan in source:
            print("Number of columns:", np.sum(np.array(scan[0].timestamp) != 0))
```

However, when I then read the .pcap files using this script (which works with the example .pcap files from the sdk), I get an error: 
```

def pcap_read_packets(
        source_file: str,
        num: int = 0  # not used in this example
) -> None:
    """Basic read packets example from pcap file. """
    # open source
    pcapMPR = pcap.PcapMultiPacketReader(source_file)
    source = pcapMPR.single_source(0)
    metadata = source.metadata

    # [doc-stag-pcap-read-packets]
    packet_format = client.PacketFormat(metadata)
    for i, packet in enumerate(source):
        print(f"Read Packet {i}")

if __name__ == "__main__":
    pcap_read_packets("C:/Users/lough/Desktop/1744315300.pcap")

pcapMPR = pcap.PcapMultiPacketReader(source_file)
  File "C:\Users\lough\AppData\Local\Programs\Python\Python310\lib\site-packages\ouster\sdk\pcap\pcap_multi_packet_reader.py", line 81, in __init__
    stats = _packet_info_stream(pcap_path, n_packets)
  File "C:\Users\lough\AppData\Local\Programs\Python\Python310\lib\site-packages\ouster\sdk\pcap\pcap.py", line 33, in _packet_info_stream
    result = _pcap.get_stream_info(path, n_packets)
RuntimeError: unknown file format
```

I am therefore guessing I am not saving the lidar packets to the .pcap file properly. 
Thanks !

loughlindd, 2025-04-14:
Hi. 

Thank you very much for your help, this was very useful. 

Just a few more questions: 
- This example requires using "private" methods from Ouster SDK prefixed with an underscore, which seems a little risky update-wise.  Are these methods expected to change in a near future or is it reasonably safe ?
- I am guessing I am probably not the only one wanting to log scans while processing them. Should it be expected that the sdk will have methods to make this easier in the future ? Or at least have an example and documentation on the website containing the steps as to how to do this ? I could only find examples logging to .pcap or processing data, but not both at once. 

Thanks

Samahu, 2025-04-14:
> This example requires using "private" methods from Ouster SDK prefixed with an underscore, which seems a little risky update-wise. Are these methods expected to change in a near future or is it reasonably safe ?

I don't see why do you consider this to be risky. If these changes in future releases you can stick to the current release or update the references if things moved around.

> I am guessing I am probably not the only one wanting to log scans while processing them. Should it be expected that the sdk will have methods to make this easier in the future ? Or at least have an example and documentation on the website

Not many people have asked for this feature to make it a priority. But yeah we may have something that makes easier in future iterations .. maybe through a callbacks/hooks or something like that.

> Or at least have an example and documentation on the website containing the steps as to how to do this ?

We are always expanding our knowledge with new examples and new content , webinars, etc. So always be on the lookout for new releases and announcements.

closed 2025-04-14 by Samahu