ouster-lidar/ouster-sdk #653: Socket binding to 0.0.0.0 attempts to parse packets from other sources

Aposhian, 2025-04-17:
**Describe the bug**
At least in my configuration, with IPv4 and no mulitcast, `udp_data_socket` binds 0.0.0.0, so then it accepts any data sent on port 7502 on any of the computers interfaces. The situation I ran into was unicast DDS discovery messages being sent to 7502 as well, but on loopback (according to this https://cyclonedds.io/docs/cyclonedds/latest/config/port_numbers.html). When these discovery messages showed up every 8 seconds, the Ouster SDK would log:

```
Unexpected udp packet length: 544
```
and then output garbage data.

Obviously, it is up to the user to make sure that ports don't conflict, but I think that the Ouster SDK could make this easier to troubleshoot by:
- binding only to the address/interface that the lidar is actually on to limit the amount of conflict
- more clearly log when packets are completely invalid (this is up to the os_sensor ROS node, not necessarily the SDK. I will make a separate ticket on ouster-ros for this).

**To Reproduce**
- Setup data to stream on port 7502 (default)
- Send UDP packets to port 7502 on loopback/localhost

**Platform (please complete the following information):**
- Ouster Sensor? OS-0
- Ouster Firmware Version? 2.5
- Programming Language? C++
- Operating System? Linux
- Machine Architecture? x64
- git commit hash? 483206f0baeffbbc11ee0883d54a651a5216563b

matthew-lidar, 2025-05-09:
Auto detecting the interface/address to bind to is rather fraught with potential issues so we would prefer to avoid doing that by default. It also doesn't really prevent receiving unexpected packets, especially since many ROS applications need discovery traffic to leave the loopback interface.

One thing we could do is allow providing an IP to bind to or a way to opt into detecting the IP to bind to. There are other applications where you might have multiple network interfaces connected to separate lidars and want to run a separate client/driver for each even if they are on the same UDP ports.

Aposhian, 2025-05-13:
I think being able to specify the binding address could be a good solution, and would probably greatly reduce the likelihood of this occurrence (even though it wouldn't eliminate it).

Another thing we could maybe do would be to capture the sender address from `recvfrom` and filter by the sender IP.

matthew-lidar, 2025-05-13:
The newer APIs in the SDK definitely filter by IP. The ROS driver likely uses different APIs so would have to implement that itself.